Xinyuan-LilyGO/T-Deck #3: Display viewing angle and mounting direction

PeterAJansen, 2023-06-16:
Hi there, 

I love the idea of this device.  But, I just tried mine out, and (a) the display has a very poor viewing angle, and (b) the viewing angle is mounted horizontally instead of vertically.  That means the only good way to view the screen is from the right, and unless you hold it at a very sharp angle, your eyes see two very different pictures.  It's driving my eyes crazy, and makes it unusable for me.  I hope you switch to a much better display on the next revision.

best wishes,
Peter

lewisxhe, 2023-07-24:
Thank you, the screen has been replaced with another brand

closed 2023-07-24 by lewisxhe

PeterAJansen, 2023-09-11:
The original purchasers should have their devices replaced.  This is not a usable product.

[image]