Xinyuan-LilyGO/T-Deck #6: T-Deck case/shell design files

mischief, 2023-08-22:
hi,

i received T-Deck recently in the mail and wanted a case.

there are design files in this git repo but i didn't have success using them. can you provide any instructions, or photos on their usage?

- T_Deck_1.stp: appears to be bottom case only, no front cover. is that expected?
- T_Deck_2.stp: i printed this, but the top cover doesn't seem to fit. there is parts of the model in the way where the keyboard/trackball is supposed to go and i didn't want to disassemble my t-deck keyboard for fear of breaking it.
- T-Deck.dxf: i can't use this file, does not work in freecad, free 3d editor for linux
- T-Deck.stp: doesn't look like a case.

i'm using freecad on gentoo linux to view the design files. it would be great if there was better support for making cases here. :smile:

ppetermann, 2023-09-19:
yeh, mine didn't fit into `T_Deck_2.stp` either - although what for me is in the way is the bottom of the keyboard not going into the top shell all the way.

derekmulcahy, 2023-10-13:
The t-deck must be partially disassembled then reassembled to make it fit in the case. The keyboard is held down by double sided tape. The case works quite well. There is a YouTube video of the process. I can find it if you can't. Good luck! The Meshtastic discord group has an #enclosures chat where various cases are being discussed.

makerdiaryid, 2023-10-29:
[image]

you can check my case for T-DECK
it have option to add RFID PN532

[T-DECK CASE](https://www.thingiverse.com/thing:6287501)

jharvey, 2023-12-11:
I've created a case in onshape. The source files can be found at the link below. 
https://cad.onshape.com/documents/b289702058d3b69984c9ded4/w/4e1c5bb6efe28ab1e1699733/e/336c687f36e879030984626b

Some key notes about this layout.
- Flat sides are intentional so it can sit on a shelf or similar surface.
- It's a bit thicker to make it more robust. Feel free to make copies and modify it. It's still reasonably small, so I'm OK with the slightly thicker nature of this case layout. 
- The front panel allows for different back panels to be installed, This back panel contains 18650 batteries. I plan to include circuits for short circuit, under voltage and a dedicated LION charger per cell. As of the time of writing this, the back panel is still a work in progress. 
- Once GPS module arrives I plan to incorporate it into this back panel. I need it before I can measure and model it. 
- I created a 3D PCB model using the Lilygo DXF and KiCAD. The components like USB jack etc. are probably not perfectly aligned. I suggest the PCB model be considered more of a rough estimate. 
- Feel free to comment to me about this case layout. I'm open to suggestions and collaboration. 

[image]
[image]
[image]
[image]

mischief, 2023-12-11:
@jharvey thanks. is there a way to edit and print this from Linux?

jharvey, 2023-12-11:
@mischief please open link and learn onshape. Onshape is in a browser and is independent of the operating system. It's kind of dependent on the browser. You should be able to make your own copy and modify that copy.

mischief, 2023-12-12:
unfortunately I do not have any interest in using a web based product like onshape for work or in using non free software that can't run on linux.

it would be great if we can share this in a public git repo in a 3d model format that can be edited and sliced for printing using free software.

zzbeakerzz, 2023-12-21:
Hey @jharvey, this is super neat! Have you had any time to continue messing with this? I actually really like the way you put all of this together. What are your thoughts of having some of this machined out of aluminum?

jharvey, 2023-12-22:
I recently got some charging boards. I'll 3D model them and incorporate them soon-ish during Christmas break. I've got overload/understand voltage protection boards coming. Those will be incorporated when I get them.

I kind of like the idea of metal, but that will likely de-tune the esp32 antenna. The Lora antenna would still be ok. But the other antenna which I don't plan to use would likely not work.

zzbeakerzz, 2023-12-24:
@jharvey well keep up the good work and consider me a subscriber! I think you have the most legit design out of the handful that are out there!

zzbeakerzz, 2024-01-08:
@jharvey I ended up having your model printed and it just arrived on Saturday. I am still waiting for the battery contactors to come in and I will get those installed and soldered together. Overall the case fits pretty nice! I had it printed in some heavy duty material so that I could modify it as needed. I ended up opening up the antenna hole to 1/4" to allow the connector to fit through (I can send you pictures if needed). I also opened up the screw head holes to 3/16" (this could have just been needed due to printing tolerances). With the material that I used for the print, I believe that the wall thickness could be reduced (40-50%?). I am also curious if a more cell phone battery shaped battery could be found (maybe 5,000 mAh? I have a few 3,000 mAh, but I like how big you went with the 3x 18650s). I only say that maybe a square LiPo could be used so that you could shrink the overall package. Overall, great job and I think that your model is my favorite to date!

jharvey, 2024-01-08:
@zzbeakerzz when did you send for print? I made some changes around Jan 1st 2024. Did those get printed? Pictures would be great. Knowing you printed makes me want to set a version in onshape so we have a record of it's status when it was printed. 

Thanks for the feedback. I'll change the sma for sure. I need to better understand the 3/16 and 50% before I make those changes. Did you get the McMaster hardware from the onshape model? Perhaps hardware is different and perhaps I should allow clearance to accommodate for a wider range of hardware. 

I've been fighting with my printer. It went so well that I've got a new printer showing up perhaps in 2 more days. This weekend end, I might have printing abilities again.

I'm picturing this will typically sit on a window sill with sunlight on the solar panel. Over time it should change the 18650's. I plan that it could also be a solar battery charger. I want to add a push button battery charger indicator, and easily removable batteries so I can hot swap and scoff off with the full batteries.

I plan for it to run with 0 to 3 batteries. If 0 battery then direct solar or plugged into wall is required. Currently it's got three purchased PCBs smashed in it. Once I know the circuit works, I plan to spin a PCB. It will look much better once I spin that PCB. For now I'd suggest only use 1 battery. If you put in a dead battery and a fully charged battery it would likely do bad things as the pack balanced out.

zzbeakerzz, 2024-01-13:
@jharvey, I shot you an email earlier this week. I am not sure if you had received it. But I did put a few photos in there. Let me know if you need any of the part numbers that I used.

jharvey, 2024-01-13:
@zzbeakerzz Yes I got the email. Sorry for not replying. Thanks for the pictures. 

I've got the SV04 printing t-beam enclosures. I'm hoping to make good progress on this this weekend. Once I get the simpler t-beam to were I'm happy with it I'll print one of these and start picking away at this design. This one will take a lot longer to print.

zzbeakerzz, 2024-01-13:
Hey!
I figured this might be easier to talk and I was not sure if this was a
valid email! I did not mean to rush a response. I honestly just wanted to
make sure that you got my message and didn't think I was ignoring you.

-Nathan

jdcrunchman, 2024-03-04:
Why doesn't this have an SNA connector for connecting to my external antenna.   I looked at the site and the photos,  but I cannot see the antenna connection.    Can you please explain why?

fonix232, 2024-04-17:
@jdcrunchman SMA connectors are large and bulky for small integrated devices. The LoRa radio uses a u.FL/IPEX MHF1 connector, as do most other ESP based boards. The MCU itself uses an integrated antenna without the option for external connectivity so you'd need to replace the ESP32-S3 with a module that has a u.FL connector.

Nonetheless, u.FL can be adapted to SMA. However it's a bit rude to litter other issues with your unrelated problem :)

skippy814, 2024-05-05:
I have been greatly enjoying your case, it's the most robust and well planned out case I have ever seen. I did notice an issue with it, however. The ON and OFF labels are in the wrong place, but I am not sure if that's due to the board region or maybe a change LilyGo may have made to the boards at one time or what.

Great fit and finish for everything though. Best I have seen!

mischief, 2024-09-18:
considering the T-Deck Plus was recently released with it's own case design that is included, i will close this. thanks for everyone's responses. unfortunately, i did not find anything satisfactory for my own use case, and it would be great for the T-Deck Plus design files to be freely available in a format that is usable in free and open source 3d modelers, but that can be a separate issue.

closed 2024-09-18 by mischief