Xinyuan-LilyGO/T-Deck #7: Missing file?

BugsTw, 2023-08-23:
“esp_vad.h: no such file or directory ”
Can’t seem to locate said missing file 
[image]

Jokymon, 2023-08-28:
This file is part of the Espressif Speech Recognition SDK (https://github.com/espressif/esp-sr/tree/master/include/esp32s3) I would expect it to exist, when you correctly installed your PlatformIO setup before building.

jcampbellco, 2023-09-06:
Hi @Jokymon, hoping to get some guidance, I'm experiencing the same issue @BugsTw is, getting an `esp_vad.h: No such file or directory` error. In my case, it's with a fresh PlatformIO install and an unmodified clone of `master`, but I receive the same error:

[image]

Based on your comment that it should be pulled in via the Speech Recognition SDK, I added a line to platformio.ini, under the `[env:T-Deck]` section `lib_deps = https://github.com/espressif/esp-sr.git`, resulting in line 39 being added:

[image]

This pulled the `esp-sr` lib in, but `UnitTest.ino` was still reporting a file not found, poking around a bit I changed the `#include` line to `#include <esp32s3/esp_vad.h>`:

[image]

This does resolve the issue of `esp_vad.h` not being found, but in turn creates 3 new errors from `esp-sr` itself:

[image]

At this point I'm a little out of my depth with the dependency resolution happening here. Is this an issue that I'm on a Windows system? I will try and replicate later from my Linux machine.

If you have any suggestions on what this might point to, especially if it's something about my configuration of PlatformIO or anything of that sort, I really appreciate it! If there's any other info I can provide or steps I should take, happy to try and report back with results. Thank you again!

jcampbellco, 2023-09-07:
@BugsTw (and anyone else who finds this issue applies to them) I was able to modify `UnitTest.ino` to remove the `esp_vad.h` dependency pretty easily, here's a diff for it (also attached as `UnitTest.patch` for convenience).

If possible any help figuring out what I've configured incorrectly on PlatformIO or any suggestions there are very much appreciated still! 

```
diff --git a/examples/UnitTest/UnitTest.ino b/examples/UnitTest/UnitTest.ino
index b21cdd8..03d4034 100644
--- a/examples/UnitTest/UnitTest.ino
+++ b/examples/UnitTest/UnitTest.ino
@@ -25,7 +25,6 @@
 #include "es7210.h"
 #include <Audio.h>
 #include <driver/i2s.h>
-#include <esp_vad.h>

 #define TOUCH_MODULES_GT911
 #include "TouchLib.h"
@@ -71,8 +70,6 @@ TFT_eSPI        tft;
 Audio           audio;
 size_t          bytes_read;
 uint8_t         status;
-int16_t         *vad_buff;
-vad_handle_t    vad_inst;
 TaskHandle_t    playHandle = NULL;
 TaskHandle_t    radioHandle = NULL;

@@ -803,34 +800,6 @@ void initBoard()
     i2s_set_pin(I2S_CH, &pin_config);
     i2s_zero_dma_buffer(I2S_CH);

-
-    vad_inst = vad_create(VAD_MODE_0);
-    vad_buff = (int16_t *)malloc(VAD_BUFFER_LENGTH * sizeof(short));
-    if (vad_buff == NULL) {
-        while (1) {
-            Serial.println("Memory allocation failed!");
-            delay(1000);
-        }
-    }
-
-    // Wait until sound is detected before continuing
-    uint32_t c = 0;
-    while (1) {
-        i2s_read(I2S_CH, (char *)vad_buff, VAD_BUFFER_LENGTH * sizeof(short), &bytes_read, portMAX_DELAY);
-        // Feed samples to the VAD process and get the result
-        vad_state_t vad_state = vad_process(vad_inst, vad_buff, VAD_SAMPLE_RATE_HZ, VAD_FRAME_LENGTH_MS);
-        if (vad_state == VAD_SPEECH) {
-            Serial.print(millis());
-            Serial.println("Speech detected");
-            c++;
-            snprintf(buf, 256, "%s:%d\n", "Speech detected", c);
-            addMessage(buf);
-        }
-        if (c >= 5)break;
-        lv_task_handler();
-        delay(5);
-    }
-
     i2s_driver_uninstall(I2S_CH);

     pinMode(BOARD_BOOT_PIN, INPUT);
```

closed 2024-10-02 by github-actions[bot]