Xinyuan-LilyGO/T-Deck #9: Keyboard interrupt

mkellner, 2023-08-26:
Is the keyboard interrupt pin connected?

Does it register high or low when a key has been pressed?

mischief, 2023-08-28:
hi,

> Is the keyboard interrupt pin connected?

yes.

i did some experiments with this. in the schematic, the pin labeled `C3 INT` is connected from GPIO8 on the esp32c3 to GPIO46 on the esp32s3.

in my fork i have a branch where i pulse GPIO8 when there's keyboard data on the esp32c3, and use that to trigger i2c reads on the esp32s3: https://github.com/mischief/Xinyuan-LilyGO-T-Deck/compare/master...mischief:Xinyuan-LilyGO-T-Deck:kbint?expand=1

> Does it register high or low when a key has been pressed?

it does whatever you program it to do on the esp32c3 side :slightly_smiling_face:

mkellner, 2023-08-28:
I am working on an SDK for esp32 devices and would like to know if it is enabled as shipped.

It is easy for beginners to program the T-Deck from the USB port, but to force them to program the keyboard esp32c3 may be a stretch too far for some.

Do we know if the interrupt pin is enabled by default on the shipping product?

mischief, 2023-08-28:
hi,

> Do we know if the interrupt pin is enabled by default on the shipping product?

no. it's not the default today.

closed 2023-08-28 by mkellner