Xinyuan-LilyGO/T-Deck #15: How to generate a simple tone on speaker

spleenware, 2023-10-07:
Can you add an example sketch on how to generate a simple tone to play on the built-in speaker (over I2S, I assume?).

technoblogy, 2023-10-08:
Here's a simple version:
````text
void beep (int freq, int duration) {
  I2S.setAllPins(7, 5, 6, 6, 6); // sckPin, fsPin, sdPin, outSdPin, inSdPin
  const int samplerate = 8000;
  int halfwave = samplerate / freq, count = 0, amplitude = 500;
  unsigned long start = millis();
  if (!I2S.begin(I2S_PHILIPS_MODE, samplerate, 16)) return; // Error
  while (millis() < start + duration) {
    if (count % halfwave == 0) amplitude = -1 * amplitude;
    I2S.write(amplitude); I2S.write(amplitude);
    count++;
  }
  I2S.end();
}
````

spleenware, 2023-10-10:
Which header files are needed, and how is I2S object constructed? (specifically for ESP32/T-Deck target)

spleenware, 2023-10-10:
Ah, thank you. I hear a beep now! :-)

aaron-924, 2024-06-07:
Is there a reason why "Play MP3" doesn't work in the UnitTest  code?

closed 2024-10-02 by github-actions[bot]