Xinyuan-LilyGO/T-Deck #23: Can't compile unit test

lazzaroevan, 2024-03-29:
Using a fresh clone of the repository, I can't build the project with unit test selected. When I do, it says that it is missing 

"*** missing SConscript file 'script\\pos_extra_script.py'
File "C:\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 180, in <module>"

If I try to comment this out, then it says

"xtensa-esp32s3-elf-g++: error: examples/UnitTest/UnitTest.ino.cpp: No such file or directory
xtensa-esp32s3-elf-g++: fatal error: no input files"

Please advise

lewisxhe, 2024-03-29:
remove this line in platformio.ini
```
extra_scripts = script\pos_extra_script.py

```

lazzaroevan, 2024-04-02:
Like I said in my first comment, when I take that line out it errors out with

"xtensa-esp32s3-elf-g++: error: examples/UnitTest/UnitTest.ino.cpp: No such file or directory
xtensa-esp32s3-elf-g++: fatal error: no input files"

lewisxhe, 2024-04-03:
In your case, platformio is not ready yet or the error will be reported after the installation is not completed. Please re-download the warehouse code and reopen it.
[image]

closed 2024-09-17 by github-actions[bot]