Xinyuan-LilyGO/T-Deck #28: New T-Deck cannot enter DFU nor gets detected by any computer as USB

SAMGARTNER, 2024-05-11:
Hello, I bought this T-Deck directly at lilygo.cc. I receive it yesterday, I have been trying to flash it but it doesn't respond. There is only a bluelight in the rear and it blinks once when I turn the unit on. I changed cables and used two computer a Macbook and a Windows laptop, none of them detects the USB connection.

As instructed, I keep pressing the scroll wheel button and turn on the unit and then release the button. I also mixed it with the reset button, but nothing happens. Looks bricked

 Is there any other procedure that I should follow?

laulin, 2024-05-18:
Hi SAMGARTNER, I had the same problem as you. I wanted to install mycropython. I did it with the esptool command ( esptool.py --port /dev/ttyACM0 --baud 460800 write_flash -z 0x0 firmware.bin). Hope you find it useful!

acidvegas, 2024-05-26:
> Hi SAMGARTNER, I had the same problem as you. I wanted to install mycropython. I did it with the esptool command ( esptool.py --port /dev/ttyACM0 --baud 460800 write_flash -z 0x0 firmware.bin). Hope you find it useful!

yeah thats not really going to help him here seeing as how its not even detected as a USB device. he wont have /dev/ttyUSB0 or /dev/ttyAMC0 

@SAMGARTNER did you happen to power this via battery ever? Or only through the usb cable? I know alot of cheap chinese LiPo batteries have the polarity wrong sometimes....

closed 2024-09-16 by github-actions[bot]