Xinyuan-LilyGO/T-Deck #33: Cant find the documentation for this board that includes any available GPIO pins

contractorwolf, 2024-06-03:
does anyone have the PDF documentation document somewhere?
thx

lewisxhe, 2024-06-04:
This board has only two available GPIOs, which are the two on the Grove socket, GPIO43 and GPIO44.

contractorwolf, 2024-06-05:
i was more trying to figure out if there was a way to turn the keyboard backlight on/off programmatically

lewisxhe, 2024-06-07:
The default firmware does not have this function. The source code of the keyboard is here. You can modify it according to your needs. https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/Keyboard_ESP32C3

contractorwolf, 2024-06-08:
how do i change the software running on the keyboard? @lewisxhe

lewisxhe, 2024-06-19:
The keyboard is controlled by ESP32C3, you need to recode C3 using a USB2TTL
[image]

closed 2024-09-16 by github-actions[bot]