Xinyuan-LilyGO/T-Deck #35: Can't Compile Unitest in TDeck from Factory - all the steps in the guide completed - Error Compilation error: 'bool enableInterrupt' redeclared as different kind of entity

cmlouro, 2024-06-06:
[image]

FQBN: esp32:esp32:esp32s3:CDCOnBoot=cdc,FlashSize=16M,PartitionScheme=app3M_fat9M_16MB,PSRAM=opi
Using board 'esp32s3' from platform in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0
Using core 'esp32' from platform in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0

"@C:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/includes" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/qio_opi/include" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\cores\\esp32" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\SPI\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\RadioLib\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\TFT_eSPI" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\FS\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\SPIFFS\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\lvgl\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\SD\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\es7210\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\Wire\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\ESP32-audioI2S\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\WiFi\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\Network\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\NetworkClientSecure\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\SD_MMC\\src" "-IC:\\Users\\carlos.louro\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\libraries\\FFat\\src" "-IC:\\Users\\carlos.louro\\Documents\\Arduino\\libraries\\TouchLib\\src" "@C:\\Users\\carlos.louro\\AppData\\Local\\Temp\\arduino\\sketches\\86632713314C0D8E57341002A4032628/build_opt.h" "@C:\\Users\\carlos.louro\\AppData\\Local\\Temp\\arduino\\sketches\\86632713314C0D8E57341002A4032628/file_opts" "C:\\Users\\carlos.louro\\AppData\\Local\\Temp\\arduino\\sketches\\86632713314C0D8E57341002A4032628\\sketch\\UnitTest.ino.cpp" -o "C:\\Users\\carlos.louro\\AppData\\Local\\Temp\\arduino\\sketches\\86632713314C0D8E57341002A4032628\\sketch\\UnitTest.ino.cpp.o"
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:111:13: error: 'bool enableInterrupt' redeclared as different kind of entity
  111 | bool        enableInterrupt = true;
      |             ^~~~~~~~~~~~~~~
In file included from C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal.h:75,
                 from C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:36,
                 from C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:19:
C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal-gpio.h:80:6: note: previous declaration 'void enableInterrupt(uint8_t)'
   80 | void enableInterrupt(uint8_t pin);
      |      ^~~~~~~~~~~~~~~
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino: In function 'void loopRadio()':
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:388:37: error: assignment of function 'void enableInterrupt(uint8_t)'
  388 |                     enableInterrupt = false;
      |                     ~~~~~~~~~~~~~~~~^~~~~~~
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:414:37: error: assignment of function 'void enableInterrupt(uint8_t)'
  414 |                     enableInterrupt = true;
      |                     ~~~~~~~~~~~~~~~~^~~~~~
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:428:33: error: assignment of function 'void enableInterrupt(uint8_t)'
  428 |                 enableInterrupt = false;
      |                 ~~~~~~~~~~~~~~~~^~~~~~~
C:\Users\carlos.louro\Downloads\T-Deck-master (1)\T-Deck-master\examples\UnitTest\UnitTest.ino:481:33: error: assignment of function 'void enableInterrupt(uint8_t)'
  481 |                 enableInterrupt = true;
      |                 ~~~~~~~~~~~~~~~~^~~~~~
Multiple libraries were found for "SD.h"
  Used: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SD
  Not used: C:\Users\carlos.louro\AppData\Local\Arduino15\libraries\SD
Using library SPI at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SPI 
Using library RadioLib at version 5.3.0 in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\RadioLib 
Using library TFT_eSPI at version 2.5.22 in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\TFT_eSPI 
Using library FS at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\FS 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SPIFFS 
Using library lvgl at version 8.3.0-dev in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\lvgl 
Using library SD at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SD 
Using library ES7210 at version 1.0 in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\es7210 
Using library Wire at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\Wire 
Using library ESP32-audioI2S-master at version 2.0.6 in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\ESP32-audioI2S 
Using library WiFi at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WiFi 
Using library Networking at version 1.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\Network 
Using library NetworkClientSecure at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\NetworkClientSecure 
Using library SD_MMC at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SD_MMC 
Using library FFat at version 2.0.0 in folder: C:\Users\carlos.louro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\FFat 
Using library TouchLib at version 0.0.2 in folder: C:\Users\carlos.louro\Documents\Arduino\libraries\TouchLib 
exit status 1

Compilation error: 'bool enableInterrupt' redeclared as different kind of entity

lewisxhe, 2024-06-07:
Downgrade esp32 core 3.0.0 to 2.0.14

aaron-924, 2024-06-07:
now that there is a major release to core 3, can we expect updates in tdeck code to support?

lewisxhe, 2024-06-07:
Not for now, because the upstream TFT_eSPI currently only supports versions below 3.0

cmlouro, 2024-06-07:
> Downgrade esp32 core 3.0.0 to 2.0.14

Thanks will do this

closed 2024-08-16 by cmlouro