Xinyuan-LilyGO/T-Deck #36: UnitTest example code shows nothing but a black screen on the t-deck

kemodaki, 2024-07-10:
i followed all the instructions on setting up the IDE for the tdeck, tried reinstalling the arduino IDE, both on linux and windows, and everything the code compiled and uploaded just fine to the t-deck, but upon uploading the code the t-deck showed nothing, the screen wouldnt even turn on, the backlight of the keyboard would flash once every reset but nothing would happen on the screen, please help

lewisxhe, 2024-07-10:
Directly write the test firmware to test whether the hardware is normal
https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/firmware_T-Deck-factory.bin

kemodaki, 2024-07-10:
> Directly write the test firmware to test whether the hardware is normal https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/firmware_T-Deck-factory.bin

yes, sorry i forgot to mention that in the original post, ive already tried that using esptool in the terminal and it works, so i dont think theres a hardware problem with the t-deck, the issue seems to be only with the arduino IDE

lewisxhe, 2024-07-10:
Which version of esp32 are you using? Versions higher than 3.0 will not work properly

kemodaki, 2024-07-10:
im using version 2.0.17

lewisxhe, 2024-07-10:
Open the serial monitor and see what messages are output

closed 2024-09-16 by github-actions[bot]