Xinyuan-LilyGO/T-Deck #38: Access IIC Pins

mattcalhoun1, 2024-07-29:
I am looking to port a private mesh communication system over to t-deck (https://chatters.io) and I really need to get an IIC connection (spi would be great too).

Looking at the back of the t-deck, the pinout, and the schematic, it doesn't look to me like there is any exposure of IIC pins on the back of the t-deck. I see 2 very nice uart connections, but not IIC.

I need to add an accurate RTC, as it's critical to the security and a few other core features of the firmware. 

Any suggestions, specifically regarding the RTC? I've ordered (not received yet) a couple of different GPS modules with battery backup. I really don't like that solution for a number of reasons, but that's my fallback option.

I'm hoping some easy I2C access is just hiding there somewhere, maybe I can pop off the keyboard and get at them or something, but it's not obvious to me.

I guess I could solder directly to the same pads the esp32 is connected to, but that seems like a bad solution as well.

Thanks

mattcalhoun1, 2024-07-29:
Btw, sorry to open this as an "issue" as it's really just a question, but seems like this is the place to ask

mattcalhoun1, 2024-07-29:
Also looking at hijacking one set of the uart pins with "SoftWire" , we'll see how that goes

lewisxhe, 2024-08-30:
Grove interface can be used as both UART and I2C

closed 2024-10-15 by github-actions[bot]