Xinyuan-LilyGO/T-Deck #41: Which board should be used for LiliGo T-keyboard

kwccoin, 2024-08-31:
For both PlatformIO and Arduino, you have only 

- LilyGo T-DIsplay
- LilyGo T-Dislay-S3

There is no T-keyboard which I believed is C3.  What should I do?

lewisxhe, 2024-08-31:
[image]

kwccoin, 2024-08-31:
Thanks.  It will be ESP32C3 Dev module.

kwccoin, 2024-08-31:
Strange ... the doc said it is C3 and even the message said it is "Keyboard C3" but I got this error:

A fatal error occurred: This chip is ESP32-S3 not ESP32-C3. Wrong --chip argument?
Failed uploading: uploading error: exit status 2

Also if I do this upload successfully, where to find the LoRa pgm/module/... which is currently running under

lewisxhe, 2024-08-31:
[image]
Need external USB2TTL to be connected to the programming port instead of TYPE-C

kwccoin, 2024-08-31:
THanks.  How to restore to this LoRA interface after I successfully upload my scratch.  It will replace it I assumed?  

I do not find in your example any LoRa interface .ino.

lewisxhe, 2024-08-31:
What is LoRa interface?

kwccoin, 2024-08-31:
It is a module that is optional but installed in my LilyGO-T keyboard.  Mine is 433MHz and the label said

LILYGO
433/470Mhz
Model: LoRa32
SX1262

I believed it is about communication.

kwccoin, 2024-08-31:
> ![image](https://private-user-images.githubusercontent.com/22990954/363386969-2e0ea049-1de0-419a-885b-88cfa707a4af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Just place order to get those cables ... For the LiliGO LoRa module, how to restore it is still awaiting your answer.  Thanks for the speedy support.

lewisxhe, 2024-08-31:
[UnitTest](https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/UnitTest) contains LoRa examples

closed 2024-10-16 by github-actions[bot]