Xinyuan-LilyGO/T-Deck #42: T-Deck Plus: possible touch screen hw issue

mverch67, 2024-08-31:
Did you notice that the T-Deck's Plus touch screen is not working as expected? Touched once it delivers I2C touch data for about ten seconds. Reason might be a defective capacitor.

lewisxhe, 2024-09-02:
There is a simple hardware test program to test the touch screen before leaving the factory. My test here is that the coordinates will only be reported after pressing. Can you provide the issue link of the report?

lewisxhe, 2024-09-02:
I have received reports from users, but I wrote a simple lvgl sliding example and there is no problem, and I checked the coordinate reporting rate of both screens is 5ms.

You can test this simple example https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/lvgl_example

lewisxhe, 2024-09-02:
I modified the touch driver logic. Here is the demo [video](https://youtu.be/ZwjJLd4vJ2g) and [firmware.factory.zip](https://github.com/user-attachments/files/16833821/firmware.factory.zip)I recorded. You can test it.

RikomagicCloudsto, 2024-09-02:
Thanks, please can you detail the changes you made to the touch driver logic or upload the source so we can incorporate the fix in all versions of T-Deck plus firmware. 
Thanks.

lewisxhe, 2024-09-02:
This is where I changed it, which involves changing LovyanGFX. If you don't want to change it, you can change the interrupt pin to -1 when initializing the screen.

[image]

mverch67, 2024-09-02:
Hi, thanks for the quick reply.
This looks like a workaround for a non-functional TOUCH_INT?
Is there a way to get TOUCH_INT working also?

lewisxhe, 2024-09-04:
Of course, you can use it by modifying the following figure
[image]

lewisxhe, 2024-09-05:
The root cause of the touch failure is that the old screen is triggered by the falling edge after touching, while the new screen is triggered by the rising edge after touching, which leads to this problem.

Although GT911 can set the interrupt trigger mode, it is invalid on the new screen. I need to communicate with the supplier to see if the trigger mode can be reset.

lewisxhe, 2024-09-10:
@mverch67  
Modify the interrupt mode of the screen according to the figure below, so that all GT911 touch screens use a unified interrupt mode, and no other places need to be modified.
[image]

lewisxhe, 2024-09-11:
The factory report does not match my test results. It is still abnormal after setting. I tested this touch screen alone and the behavior is normal, but the interrupt setting does not seem to have any effect after installing it on T-Deck. I will continue to investigate.

The current solution is to remove the monitoring interrupt first and use the polling register, so that it can be applied to T-Deck and T-Deck-Plus
@mverch67

Achan1125, 2024-09-11:
> I modified the touch driver logic. Here is the demo [video](https://youtu.be/ZwjJLd4vJ2g) and [firmware.factory.zip](https://github.com/user-attachments/files/16833821/firmware.factory.zip) I recorded. You can test it.

Thanks!!!!  I use the other firmware for my 2 T-Deck plus and they all dead!(reboot loop)
( https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/firmware )  <--- Brick for my T-Deck+
but this firmware can work!

mverch67, 2024-09-11:
I figured out, that when touching the screen the T-Deck is pulling the TOUCH_INT GPIO(16) to _low_, but the T-Deck Plus is pulling it _high_. So my current solution to support both variants in a single firmware is to read the GPIO(16) during startup when it is idle and remember this value. Thus, later I can use this value for the TOUCH_INT processing to know if the screen is touched or not.

lewisxhe, 2024-09-13:
@mverch67 I was going in the wrong direction before, it doesn't need to be that complicated
GT911 saves the touch configuration by default, and it only needs to be set once. I reset all the interrupt triggering methods of the touch screen in the factory program of T-Deck, so that the original settings can be overwritten. The mesh firmware does not need to be modified.

The board that had problems before can be flashed with the [UnitTest](https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/T-Deck_UnitTest_240913.bin) firmware first, and then the meshtastic firmware can be flashed again after the startup is completed, and it can be used normally.

mverch67, 2024-09-30:
I personally don't have have T-Deck plus, but there are reports that your proposal worked. So I think this issue can be closed.

closed 2024-10-01 by lewisxhe