Xinyuan-LilyGO/T-Deck #45: Lorawan

RobbyGit, 2024-09-15:
Hi 

Is anyone working on getting the t-deck to work with Lorawan and not lora.

lewisxhe, 2024-09-18:
If you are familiar with the use of Lora, then you can see here [LoRaWAN](https://github.com/jgromes/RadioLib/tree/master/examples/LoRaWAN)

RobbyGit, 2024-09-18:
Nope this is not the right chip That example is SX1278 the T-deck used SX1262

Initialise the radio
Initialise radio failed - ERR_CHIP_NOT_FOUND (-2)

lewisxhe, 2024-09-19:
You are right. The link you sent is just an example of RadioLib. You need to modify it according to t-deck to the same model and pin definition as t-deck.

lewisxhe, 2024-09-19:
I took the time to add a lorawan example today.
https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/LoRaWAN_Starter

RobbyGit, 2024-09-19:
ok great will have a look. I tied that and it didn't seem to find the CHIP. Perhaps I had the pins wrong. Will feedback but thank you so very much. This will open up a whole new area for the T_deck

RobbyGit, 2024-09-19:
ok I was right ...Your example has the sameproblem mine does also.. 
Setup ... 
Initialise the radio
Initialise radio failed - ERR_CHIP_NOT_FOUND (-2)

I think this has to do with .
SX1262 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

I dont thinks the pins are right

lewisxhe, 2024-09-19:
If so, the one you bought does not have LoRa function. Please check your motherboard and order.

RobbyGit, 2024-09-19:
I have the T-Deck(ESP32S3) with the SX1262 Lora chip

lewisxhe, 2024-09-19:
Please take a photo of the motherboard. It is the back, there should be a module with SX1262 on it.

RobbyGit, 2024-09-19:
Can you share you platfomio.ini.. maybe mine is wrong

RobbyGit, 2024-09-19:
[image]

lewisxhe, 2024-09-19:
Did you use the sketch I uploaded or did you just modify the pins?

lewisxhe, 2024-09-19:
If you use the sketch I provided, there should be no error, because this is the sketch I have tested.

Try this sketch https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/UnitTest or use the binary file directly https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/T-Deck_UnitTest_240913.bin

RobbyGit, 2024-09-19:
Thank you Lewis, Seem when i updated the firmware to T_DDeck_UnitTest_240913.bin and then rebuilt the code after a clean ... it works... Wow!! thank you so much for the time you spent on it . I am going to build a small message app and will let you know, maybe include it in the examples. Will let you know

RobbyGit, 2024-09-19:
Work perfectly. ;-)

closed 2024-09-19 by RobbyGit