Xinyuan-LilyGO/T-Deck #46: Clear Developer Documentation?

rondotdll, 2024-09-16:
Is there a place to find clear, written out developer documentation explaining what libraries are intended to be used with the T-Deck, how to use specific components, how to perform specific functions, etc? Unless I'm missing something, it seems the only way to learn to develop custom firmware / programs for the device is by reversing the (sparsely documented) examaple files?

I'm attempting to develop my own GUI library and would like to use the trackball for basic traversal, and have no need for pointer functions. How can I just get the raw primitive trackball input? (up, down, left, or right)?

lewisxhe, 2024-09-18:
Sorry, no, [UnitTest](https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/examples/UnitTest/UnitTest.ino) contains all the hardware drivers and test methods for this board.

RobbyGit, 2024-09-18:
Its a mess.. There are librarys missing and examples in the github. Its a total mess

closed 2024-11-02 by github-actions[bot]