Xinyuan-LilyGO/T-Deck #52: T-Deck Plus GPS example?

richonguzman, 2024-10-03:
I cant find any info about how to activate, read or do anything to make GPS work

any info about model, brand? 

is it always powered?

lewisxhe, 2024-10-04:
This example is compatible with both u-blox and l76k GPS. T-deck and t-deckplus are exactly the same, except for an additional GPS module

https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/examples/GPSShield/GPSShield.ino

richonguzman, 2024-10-04:
is this example ment to activate into NMEA and then only use the SerialGPS.read(); in the Loop?

or should it run the complete configuration process of the GPS each time I turn on the board?

lewisxhe, 2024-10-04:
1. is this example ment to activate into NMEA and then only use the SerialGPS.read(); in the Loop?
> Yes

2. or should it run the complete configuration process of the GPS each time I turn on the board?

> No, I wrote the configuration process only to make it compatible with both GPS modules, and to restore the factory configuration of the device that has been flashed with meshtastic firmware, otherwise it will not be able to obtain NMEA and baud rate correctly. If you don't use meshtastic, the default baud rate of the ublox version is 38400, and the L76K version is 9600

richonguzman, 2024-10-04:
now you confuse me: the T-Deck Plus has only the L76K module, right? so for "non" meshtastic code I should use 9600, right?

lewisxhe, 2024-10-04:
T-Deck Plus has two GPS versions,
One is UBlox-M10Q GPS, using 38400 baud rate
The second is Quectel L76K, using 9600 baud rate

richonguzman, 2024-10-04:
got it, so I must test for which one works without opening the case.  Or can I identify them only by looking it? or how it answer to the different BAUD_SPEEDs?

lewisxhe, 2024-10-04:
Run https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/examples/GPSShield/GPSShield.ino and there will be a serial prompt

richonguzman, 2024-10-04:
awesome!

thanks a lot!! will add this soon to the Growing LoRa APRS World ;)

lewisxhe, 2024-10-04:
LoRa_APRS_Tracker ? OK. Thanks for your contribution.

Remember that T-Deck-Plus has two GPS versions! Remember to make them compatible, otherwise the fixed 9600 baud rate will cause the UBlox version to be unable to read GPS data

richonguzman, 2024-10-04:
tomorrow will test the code to see:
1) T-Deck Plus GPS identifier
2) T-Deck Plus GPS change to NMEA (depending on Gps type)
3) Two Versions/Environments for both GPS versions of T-Deck Plus

richonguzman, 2024-10-11:
a little modification to be runned on VSCODE and T-deck worked fine (not L76K but M10Q was initialized and stayed fine for later usage on LoRa APRS Tracker Firmware)

should I share here the code or should I upload to github for others to see?

lewisxhe, 2024-10-12:
Hi @richonguzman  

You can submit a simple example as a PR and put it in the examples of this repository.

closed 2024-11-25 by github-actions[bot]

xukashi, 2025-03-19:
Unfortunately, the touchscreen and GPS don't work. I found the following information about the GPS:

UBlox MIA-M10Q GNSS Module
UBlox Module using baud rate 38400
GPIO44 for GPS RX
GPIO43 for GPS TX