Xinyuan-LilyGO/T-Deck #54: T-Deck Touchscreen input flipped

THEGxHOxST, 2024-10-06:
I'm having the issue where the touch inputs are flipped from left to right, as in when I touch the right it detects it on the left and via versa

lewisxhe, 2024-10-07:
Need to swap XY axis, please refer to the example
https://github.com/Xinyuan-LilyGO/T-Deck/blob/01feb0961ec1b1d542b70d9655619090bf4cee81/examples/Touchpad/Touchpad.ino#L42

THEGxHOxST, 2024-10-18:
Forgive me, I'm trying to understand how to use this code, is this to fix it within something I made or is it, like what I'm dealing with, where every image I download and flash to this T deck is flipped. If so, how do I use that code.

THEGxHOxST, 2024-10-23:
I have figured out how to compile and upload the Unitest with the inputs correct, however it doesn't effect any other firmware I flash.

closed 2024-10-25 by THEGxHOxST