Xinyuan-LilyGO/T-Deck #55: T-Deck Plus Touch not working

mertensm72, 2024-10-07:
DOA of one of my new T-Decks: Display is functional but touch function does not work.

Is there anything I can do, or do I have to return the device?

lewisxhe, 2024-10-07:
Write the factory test file for testing. If you still cannot touch, take a video and send it to the dealer.

https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/T-Deck_UnitTest_241004.bin

mertensm72, 2024-10-07:
I already checked that out. The flash file does not start on my T-Deck Plus
Do you have a Manual for flashing the UnitTest to the T-Deck Plus?

lewisxhe, 2024-10-07:
You can check this for the writing method, which is basically the same.
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware

If you cannot write, turn the switch to OFF, then press and hold the button in the middle of the trackball, press it, don't release it, and then turn the switch to ON. At this time, it is in download mode, click Upload, wait for the writing to complete, and press the RST button to restart the device after the writing is completed.

lewisxhe, 2024-10-07:
The write start address is 0, not 0x1000

mertensm72, 2024-10-07:
That worked. But there is no test for touch. Touch also does not work in this testing firmware. I think it's defect

lewisxhe, 2024-10-07:
Shoot a test video and provide it to the dealer, they will process it

mertensm72, 2024-10-07:
I will do so. The dealer is lilygo.cc ;-)

mertensm72, 2024-10-08:
Lewis, would you mind to talk to your sales colleagues, why they don't answer my rma mail request?

lewisxhe, 2024-10-08:
I haven't received their feedback yet. You just need to let them know. Please be patient.

closed 2024-10-20 by mertensm72