Xinyuan-LilyGO/T-Deck #61: How backlight works on t-deck plus?

bettio, 2024-10-28:
Display backlight (gpio 42) is working on regular t-deck using LEDC PWM , but same code doesn't work with the Plus model.
How backlight can be enabled with esp-idf APIs? Should I use any specific PWM frequency or duty cycle?

lewisxhe, 2024-10-29:
https://github.com/Xinyuan-LilyGO/T-Deck/blob/c43766f13c61b786777405e17fd9e6d6c8bed86c/examples/UnitTest/UnitTest.ino#L278

bettio, 2024-10-29:
Thanks for the reply, I have 2 more questions: does `setBrightness` works same way on both Plus and non Plus?
Furthermore, I tried to translate it to regular esp-idf code, with no success:
```c
#define BOARD_BL_PIN 42

void setBrightness(uint8_t value)
{
    gpio_set_direction(BOARD_BL_PIN, GPIO_MODE_OUTPUT);

    static uint8_t level = 0;
    static uint8_t steps = 16;
    if (value == 0) {
        gpio_set_level(BOARD_BL_PIN, 0);
        vTaskDelay(3 / portTICK_PERIOD_MS);
        level = 0;
        return;
    }
    if (level == 0) {
        gpio_set_level(BOARD_BL_PIN, 1);
        level = steps;
        delayMicroseconds(30);
    }
    int from = steps - level;
    int to = steps - value;
    int num = (steps + to - from) % steps;
    for (int i = 0; i < num; i++) {
        gpio_set_level(BOARD_BL_PIN, 0);
        gpio_set_level(BOARD_BL_PIN, 1);
    }
    level = value;
}

#define NOP() asm volatile ("nop")

unsigned long IRAM_ATTR micros()
{
    return (unsigned long) (esp_timer_get_time());
}

void IRAM_ATTR delayMicroseconds(uint32_t us)
{
    uint32_t m = micros();
    if(us){
        uint32_t e = (m + us);
        if(m > e){ //overflow
            while(micros() > e){
                NOP();
            }
        }
        while(micros() < e){
            NOP();
        }
    }
}
```

am I missing anything?

lewisxhe, 2024-10-30:
Plus and non-Plus are the same, only the shell and screen are different

I tested it and there was no problem at all
```c
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"
#include "esp_timer.h"
#include <rom/ets_sys.h>
#define BOARD_BL_PIN 42
#define BOARD_POWERON 10

void setBrightness(uint8_t value)
{
    static uint8_t level = 0;
    static uint8_t steps = 16;
    if (value == 0)
    {
        gpio_set_level((gpio_num_t)BOARD_BL_PIN, 0);
        vTaskDelay(pdMS_TO_TICKS(3));
        level = 0;
        return;
    }
    if (level == 0)
    {
        gpio_set_level((gpio_num_t)BOARD_BL_PIN, 1);
        level = steps;
        ets_delay_us(30);
    }
    int from = steps - level;
    int to = steps - value;
    int num = (steps + to - from) % steps;
    for (int i = 0; i < num; i++)
    {
        gpio_set_level((gpio_num_t)BOARD_BL_PIN, 0);
        gpio_set_level((gpio_num_t)BOARD_BL_PIN, 1);
    }
    level = value;
}

void app_main(void)
{

    gpio_config_t config;
    memset(&config, 0, sizeof(config));
    config.pin_bit_mask = 1ULL << BOARD_POWERON;
    config.mode = GPIO_MODE_OUTPUT;
    config.pull_up_en = GPIO_PULLUP_DISABLE;
    config.pull_down_en = GPIO_PULLDOWN_DISABLE;
    config.intr_type = GPIO_INTR_DISABLE;
    ESP_ERROR_CHECK(gpio_config(&config));
    gpio_set_level((gpio_num_t)BOARD_POWERON, 1);

    gpio_config_t bl_pin;
    memset(&bl_pin, 0, sizeof(bl_pin));
    bl_pin.pin_bit_mask = 1ULL << BOARD_BL_PIN;
    bl_pin.mode = GPIO_MODE_OUTPUT;
    bl_pin.pull_up_en = GPIO_PULLUP_DISABLE;
    bl_pin.pull_down_en = GPIO_PULLDOWN_DISABLE;
    bl_pin.intr_type = GPIO_INTR_DISABLE;
    ESP_ERROR_CHECK(gpio_config(&bl_pin));

    uint8_t level = 0;
    while (1)
    {
        setBrightness(level++);
        level %= 16;
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

```

bettio, 2024-10-30:
Now it works, but there is something that I don't understand yet: how gpio 10 is involved here?
Also it looks required only for the non plus one.

lewisxhe, 2024-11-01:
https://github.com/Xinyuan-LilyGO/T-Deck/blob/c43766f13c61b786777405e17fd9e6d6c8bed86c/examples/UnitTest/UnitTest.ino#L832

closed 2024-12-16 by github-actions[bot]