Xinyuan-LilyGO/T-Deck #62: Documentation - Can't Enter Download Mode

AlexBestoso, 2024-11-06:
I've been working with the T-deck for a while and have gotten into a great many boot loops.

Holding down the scroll ball and restarting as described enters me into an invalid upload mode. 

I was finally able to get passed this issue by connecting gpio 46 to ground and then repeating the process involving the trackball.

GPIO 46 was the second from the left on the bottom row of esp pinouts.

The above outlined process fixed my issues resulting from serial data corruption due to bad code and should be mentioned in the documentation for trouble shooting flashing issues.

[image]

lewisxhe, 2024-11-07:
Thanks for the report. This problem is caused by GPIO46 being pulled up to a high level by esp32c3, which makes it impossible to enter USB download mode. The solution is to remove the resistor shown in the figure below.
[image]

AlexBestoso, 2024-12-19:
I'm not sure if removing the resistor will work or not. 

The issue I was having was do to serial corruption due to poor software being flashed. 

Which prevented my linux kernel from properly registering the esp device.

I've not tested removing the resistor; but I know for certain that connecting the mentioned pin to ground will work as describe for my particular issue.

This fix here isn't properly documented, so I made this documentation request.

closed 2025-02-04 by github-actions[bot]