Xinyuan-LilyGO/T-Deck #64: T-Deck + cannot connect via bluetooth

wq7w, 2024-11-13:
Does the t-deck + support bluetooth connection from the meshtastic app? When I am on the app and click add device I am not able to find the t-deck +

lewisxhe, 2024-11-14:
Have you written the meshtastic firmware? The bluetooth name should be meshtastic-xxxx

ricardo-maestas, 2024-11-20:
This might be because WiFi is enabled on the device. Meshtastic docs indicate these features are mutually exclusive. https://meshtastic.org/docs/configuration/radio/network/

joyel24, 2024-12-16:
@wq7w If you're using the meshtastic firmware with the TFT UI, you actually won't be able to use BT anymore connected to the smartphone app.

For the moment, fancy UI needs BT API to work.

If you flash the standard UI you'll be able to connect it to your phone I guess, but you'll lose all TFT UI advantages...

KitWalker022, 2025-01-17:
So I am understanding this correctly , I can not connect to my mobile data on a smartphone when I am away from my wifi networks? I can only connect via USB tethering if the phone supports this?
The TFT UI is pretty cool, but such a disappointment not to support BT. Will this be supported in the future and will the TFT firmware become an official release again.
I just installed it today and It took a bit of time for me to get it up and running, I really like it. I find the serial interface on my W10 pc very buggy no matter what esp32 device I have plugged in. The BT is so much more stable when programming config changes.
If anyone can answer the questions I asked it would be greatly appreciated as I am new to this world of microcontrollers and the mesh and dont have all the technical aspects understood yet.
Best regards
KIT

KitWalker022, 2025-01-17:
> So if I am understanding this correctly , I can not connect to my mobile data on a smartphone when I am away from my wifi networks? I can only connect via USB tethering if the phone supports this? The TFT UI is pretty cool, but such a disappointment not to support BT. Will this be supported in the future and will the TFT firmware become an official release again. I just installed it today and It took a bit of time for me to get it up and running, I really like it. I find the serial interface on my W10 pc very buggy no matter what esp32 device I have plugged in. The BT is so much more stable when programming config changes. If anyone can answer the questions I asked it would be greatly appreciated as I am new to this world of microcontrollers and the mesh and dont have all the technical aspects understood yet. Best regards KIT

I have sorted the lack of BT connectivity, as per joyel24, I used my phones hot spot connection and put the details in the NETWORK TAB AS FOLLOWS >WIFI=ENABLED>SSID=(Phones hotspot network name)>PASSWORD=(Phones hotspot password)> transmit config to device.
To connect the phone from the app I then went to the connection menu as per for a BT connection and then connected to the IP address that was shown there. From this point I could program Tdeck and use the phone as an interface to the device as required.
I hope this helps someone else. Thanks again.
Best regards
Kit

joyel24, 2025-01-19:
I did not specify but of course I also configured the T-deck with my phone access point so if I need to use a qr code for example I do it this way.
In the screenshot I did, T-Deck is connected to my phone's wifi, not my home AP, but both works.

lambdaknight, 2025-02-24:
@joyel24 What would it take to add the BT API to the fancy UI? Is it relatively straight forward for someone who knows what they're doing?

anarcat, 2025-03-26:
> Since not a long time ago, there is a BT programming mode in it, but you still can't use UI and BT at the same time.
> Build your own 2.6 firmware with MUI or wait because I guess it's gonna be a public release in 2 days 😉

I've flashed the "release candidate" meshtastic firmware two days ago, but still can't connect to bluetoth while the UI is on, am I doing something wrong?

I've also tried enabling WiFi, but the meshtastic app can't connect to the device: it doesn't respond on any port, although it pings at its assigned IP address.

mverch67, 2025-03-26:
> I've flashed the "release candidate" meshtastic firmware two days ago, but still can't connect to bluetoth while the UI is on, am I doing something wrong?

You'd need to go into BT programming mode (either press the meshtastic logo at the boot screen or in the setting->reboot) then you can do the programming via bluetooth. Press the bluetooth button again to boot back into MUI.

joyel24, 2025-03-26:
I'm 1 hour too late but yes, this is it. Manuel is too fast ! 
@anarcat There is documentation for Meshtastic UI now:
https://meshtastic.org/docs/software/meshtastic-ui/

closed 2025-05-10 by github-actions[bot]