Xinyuan-LilyGO/T-Deck #66: No gps on the t-deck plus

predictitai, 2024-11-20:
I have bought a T-Deck plus and tried to get the  GPSShield example to work. I notice that the initialization during setup works as intended but i don't receive any location information. All i receive is 'INVALID' and i see 0 satellites are detected. I see locations on my mobile phone so pretty sure there is GPS signal here.

I thought that it had something to do with the radio signal or antenna so i bought a second T-Deck but observed similar issues.
What could I test to see whether there is something wrong with my device or signal?

lewisxhe, 2024-11-21:
If initialization is successful, then there is no problem with communication

1. Is the device placed outdoors?

2. Is the GPS antenna connected reliably?

3. Have you tried to replace the GPS antenna?

predictitai, 2024-11-28:
1. Yes. I get time information meaning that at least it is able to receive a signal.
2. Check that and yes it is connected reliably.
3. I have used a different T-Deck plus unit as I have two with similar results. Change of both being defect is small i suppose.

When i read out the raw commands with the following code:
void loop()
{

    while (SerialGPS.available()) {
        char c = SerialGPS.read();
        Serial.print(c);  // Debugging raw NMEA data
        gps.encode(c);    // Feed character to TinyGPS++
    }

    Serial.print("Satellites: ");
    Serial.println(gps.satellites.value());

    Serial.print("HDOP: ");
    Serial.println(gps.hdop.value());

    delay(1000);
}
I receive the following info:

16:17:29.873 > $GNRMC,,V,,,,,,,,,,N,V*37
16:17:29.874 > $GNVTG,,,,,,,,,N*2E
16:17:29.875 > $GNGGA,,,,,,0,00,99.99,,,,,,*56
16:17:29.876 > $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
16:17:29.878 > $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
16:17:29.880 > $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,4*36
16:17:29.881 > $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,5*37
16:17:29.883 > $GPGSV,1,1,00,0*65
16:17:29.884 > $GAGSV,1,1,00,0*74
16:17:29.884 > $GBGSV,1,1,00,0*77
16:17:29.885 > $GQGSV,1,1,00,0*64
16:17:29.885 > $GNGLL,,,,,,V,N*7A
16:17:29.886 > Satellites: 0
16:17:29.886 > HDOP: 9999

Any advice or suggestions? I will go outside today and double validate.

predictitai, 2024-11-28:
uint8_t cfg_clear1[] = {0xB5, 0x62, 0x06, 0x09, 0x0D, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x1C, 0xA2};
    
    delay(100); // Allow module to stabilize
    SerialGPS.write(cfg_clear1, sizeof(cfg_clear1));
    if (getAck(buffer, 256, 0x05, 0x01)) {
        Serial.println("Get ack successes!");
    }
   Ack does not match for cfg_clear1

lewisxhe, 2024-11-29:
If they do not match, then you should be using the L76K GPS model.

lewisxhe, 2024-11-29:
> 1. Yes. I get time information meaning that at least it is able to receive a signal.
> 2. Check that and yes it is connected reliably.
> 3. I have used a different T-Deck plus unit as I have two with [...]

If NMEA sentences are continuously output, it means that the GPS is working. You need to place the device outdoors for testing.

closed 2025-01-12 by github-actions[bot]