Xinyuan-LilyGO/T-Deck #71: T-deck trackball not working?

asherevan, 2025-01-11:
I am trying to read from the trackball in micropython, but the down direction seems to not work. 
I am using this code:

```py
from machine import Pin

tb_up = Pin(3, mode=Pin.IN, pull=Pin.PULL_UP)
tb_down = Pin(15, mode=Pin.IN, pull=Pin.PULL_UP)
tb_left = Pin(1, mode=Pin.IN, pull=Pin.PULL_UP)
tb_right = Pin(2, mode=Pin.IN, pull=Pin.PULL_UP)
tb_click = Pin(0, mode=Pin.IN, pull=Pin.PULL_UP)
tb_up_value = tb_up.value()
tb_down_value = tb_down.value()
tb_left_value = tb_left.value()
tb_right_value = tb_right.value()

def left():
    global tb_left_value
    if tb_left.value() != tb_left_value:
        tb_left_value = tb_left.value()
        return True
    else:
        return False

def right():
    global tb_right_value
    if tb_right.value() != tb_right_value:
        tb_right_value = tb_right.value()
        return True
    else:
        return False

def up():
    global tb_up_value
    if tb_up.value() != tb_up_value:
        tb_up_value = tb_up.value()
        return True
    else:
        return False

def down():
    global tb_down_value
    if tb_down.value() != tb_down_value:
        tb_down_value = tb_down.value()
        return True
    else:
        return False

def pressed():
    return not tb_click.value()

while True:
    time.sleep(0.5)
    print('Trackball up:', up(), " Trackball down:", down(), "Trackball right:", right(), " Trackball left:", left(), "Pressed:", pressed())
```

Any ideas?

Also it would be great if you uploaded micropython examples for everything.

asherevan, 2025-01-11:
Also I cannot get the SD card to work.

lewisxhe, 2025-01-13:
You forgot to set GPIO10 high.

asherevan, 2025-01-13:
Oh. Sorry forgot to include that in the code. Updated.

asherevan, 2025-02-06:
Is there a way to fix the SD card problem?

lewisxhe, 2025-02-06:
What happened to the SD card? I thought you had no problem.

asherevan, 2025-02-06:
Oh. Sorry. I had added that in a later comment.
The micro SD card will not mount. I am using a SanDisk 32 GB MicroSD card and using this code:

```python
import machine
import os
import sdcard

sdspi=machine.SoftSPI(-1, sck=machine.Pin(40), mosi=machine.Pin(41), miso=machine.Pin(38, machine.Pin.IN))
sd = sdcard.SDCard(sdspi, machine.Pin(39))
os.mount(sd, '/sd')
print('SD card mounted.')
```

But it randomly returns one of "OSError: no response from SD card",  "OSError: timeout waiting for v2 card", or "OSError: [Errno 19] ENODEV" error.

lewisxhe, 2025-02-07:
The SPI bus is shared by the screen and LoRa. You need to set the SPI CS of the screen and LoRa to HIGH, then perform SD initialization, and set GPIO10 to HIGH. Like this example
https://github.com/Xinyuan-LilyGO/T-Deck/blob/057ad4480ad4995c773434e0aa35a412e44d1b2e/examples/SD/SD.ino#L196

asherevan, 2025-02-07:
Like this?

```python
import machine
from machine import Pin, SoftSPI
import sdcard
import os

pwr = Pin(10)
pwr.on()

sdCS = Pin(39)
screenCS = Pin(12)
loraCS = Pin(9)

screenCS.on()
sdCS.on()
loraCS.on()

MISO = Pin(38, Pin.IN, Pin.PULL_UP)
sdspi=machine.SoftSPI(-1, sck=machine.Pin(40), mosi=machine.Pin(41), miso=MISO)
sd = sdcard.SDCard(sdspi, sdCS)
os.mount(sd, '/sd')
```

lewisxhe, 2025-02-07:
Yes, you try it.

asherevan, 2025-02-07:
I tried it but It still doesn't work.

lewisxhe, 2025-02-08:
Maybe you can try the Arduino code first to test whether the SD card is available. I am not familiar with Micropython.

asherevan, 2025-03-10:
Sorry for the late reply.
I tried the unit test and it seemed to work.

asherevan, 2025-03-15:
I believe that it is a problem with the hardware. Thank you for your help!

closed 2025-03-15 by asherevan