Xinyuan-LilyGO/T-Deck #75: LoRa with esp-idf

AkaBkn, 2025-02-08:
I'm currently trying to get the LoRa working. I tried those two libraries together with esp-idf:
- https://github.com/nopnop2002/esp-idf-sx126x
- https://github.com/jgromes/RadioLib

Both giving me SPI errors, or the SX1262 chip behaves strange. Is there something to consider when trying to use those libraries with esp-idf?
Are there some differences between HPD16A and SX1262?

LoRa is working fine with different firmwares.

I tried different things with no luck so far, the first couple of commands over SPI seem to be fine, but then errors occur. here is an example output of the first library:

`
I (277) main_task: Calling app_main()
I (277) RA01S: CONFIG_MISO_GPIO=38
I (277) RA01S: CONFIG_MOSI_GPIO=41
I (277) RA01S: CONFIG_SCLK_GPIO=40
I (287) RA01S: CONFIG_NSS_GPIO=9
I (287) RA01S: CONFIG_RST_GPIO=17
I (287) RA01S: CONFIG_BUSY_GPIO=13
I (287) RA01S: CONFIG_TXEN_GPIO=-1
I (297) RA01S: CONFIG_RXEN_GPIO=-1
I (297) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (307) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (317) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (327) RA01S: spi_bus_initialize=0
I (327) RA01S: spi_bus_add_device=0
I (327) MAIN: Frequency is 866MHz
W (337) MAIN: Enable TCXO
I (337) MAIN: LoRa Begin
I (377) RA01S: ReadRegister: REG=0x740
I (377) RA01S: DataIn:14
I (377) RA01S: DataIn:24
I (377) RA01S: syncWord=0x1424
I (377) RA01S: SX126x installed
I (377) RA01S: WriteCommand: CMD=0x80 [1]
I (387) RA01S: 00 --> aa
W (387) RA01S: WriteCommand2 status=0a retry=1
I (387) RA01S: WriteCommand: CMD=0x80 [1]
I (397) RA01S: 00 --> a2
I (397) RA01S: WriteCommand: CMD=0x9d [1]
I (397) RA01S: 01 --> 00
W (407) RA01S: WriteCommand2 status=ff retry=1
I (407) RA01S: WriteCommand: CMD=0x9d [1]
I (407) RA01S: 01 --> aa
W (417) RA01S: WriteCommand2 status=0a retry=2
I (417) RA01S: WriteCommand: CMD=0x9d [1]
I (417) RA01S: 01 --> a8
W (427) RA01S: WriteCommand2 status=08 retry=3
I (427) RA01S: WriteCommand: CMD=0x9d [1]
I (427) RA01S: 01 --> a2
I (437) RA01S: tcxoVoltage=1.600000
I (437) RA01S: WriteCommand: CMD=0x97 [4]
I (437) RA01S: 00 --> a2
I (437) RA01S: 00 --> 00
W (447) RA01S: WriteCommand2 status=ff retry=1
I (447) RA01S: WriteCommand: CMD=0x97 [4]
I (447) RA01S: 00 --> aa
W (457) RA01S: WriteCommand2 status=0a retry=2
I (457) RA01S: WriteCommand: CMD=0x97 [4]
I (467) RA01S: 00 --> a2
I (467) RA01S: 00 --> a2
I (467) RA01S: 01 --> a2
I (467) RA01S: 40 --> 00
W (477) RA01S: WriteCommand2 status=ff retry=3
I (477) RA01S: WriteCommand: CMD=0x97 [4]
I (477) RA01S: 00 --> aa
W (477) RA01S: WriteCommand2 status=0a retry=4
I (487) RA01S: WriteCommand: CMD=0x97 [4]
I (487) RA01S: 00 --> a2
I (487) RA01S: 00 --> a2
I (497) RA01S: 01 --> a2
I (497) RA01S: 40 --> 00
W (497) RA01S: WriteCommand2 status=ff retry=5
I (507) RA01S: WriteCommand: CMD=0x97 [4]
I (507) RA01S: 00 --> aa
W (507) RA01S: WriteCommand2 status=0a retry=6
I (517) RA01S: WriteCommand: CMD=0x97 [4]
I (517) RA01S: 00 --> a2
I (517) RA01S: 00 --> a2
I (517) RA01S: 01 --> a2
I (527) RA01S: 40 --> 00
W (527) RA01S: WriteCommand2 status=ff retry=7
I (527) RA01S: WriteCommand: CMD=0x97 [4]
I (537) RA01S: 00 --> aa
W (537) RA01S: WriteCommand2 status=0a retry=8
I (537) RA01S: WriteCommand: CMD=0x97 [4]
I (547) RA01S: 00 --> a2
I (547) RA01S: 00 --> a2
I (547) RA01S: 01 --> a2
I (547) RA01S: 40 --> 00
W (557) RA01S: WriteCommand2 status=ff retry=9
E (557) RA01S: SPI Transaction error:0xff
`

What are the correct settings? LDO, TXO

lewisxhe, 2025-02-10:
You didn't tell me which board it was.

AkaBkn, 2025-02-10:
I'm developing for the T-Deck Plus.

lewisxhe, 2025-02-10:
1. Is GPIO10 set to high level?
2. Is CS set to high level for shared SPI bus, screen, SD card?

lewisxhe, 2025-02-10:
There are many open source examples of t-deck in the community, such as this one https://github.com/ByteWelder/Tactility

AkaBkn, 2025-02-10:
1. Yes, GPIO10 is HIGH (but thats `BOARD_POWERON`)
2. Yes, CS for RADIO is LOW during SPI transaction, otherwise all CS (radio, screen, sdcard) are HIGH
3. There are some open source examples, but I couldn't find one using LoRa on esp-idf. The one you sent doesn't even use LoRa
4. Is this corect?
```
I (277) RA01S: CONFIG_MISO_GPIO=38
I (277) RA01S: CONFIG_MOSI_GPIO=41
I (277) RA01S: CONFIG_SCLK_GPIO=40
I (287) RA01S: CONFIG_NSS_GPIO=9
I (287) RA01S: CONFIG_RST_GPIO=17
I (287) RA01S: CONFIG_BUSY_GPIO=13
```
5. Are there some differences between HPD16A and SX1262?

lewisxhe, 2025-02-10:
HPD16A Which one do you mean? Any pictures?
You configured the IO correctly, but you didn't configure DIO1 = 45

AkaBkn, 2025-02-10:
HPDA16A: I have no idea, but that is the name in the schematic (https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/schematic/schematic.pdf)

DIO1: As far as i know, its only used when configured by the radio, it can have different purposes like IRQ. But it should not be needed for basic usage, am I right?

lewisxhe, 2025-02-10:
T-deck only has SX1262 version. TCXO 3.3V, LDO does not need to be set.

lewisxhe, 2025-02-10:
There is nothing special to note, everything is normal
[image]

lewisxhe, 2025-02-10:
```c
/* The example of ESP-IDF
 *
 * This sample code is in the public domain.
 */

#include <stdio.h>
#include <inttypes.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include <driver/gpio.h>
#include "ra01s.h"

static const char *TAG = "MAIN";

#if CONFIG_SENDER
void task_tx(void *pvParameters)
{
	ESP_LOGI(pcTaskGetName(NULL), "Start");
	uint8_t buf[256]; // Maximum Payload size of SX1261/62/68 is 255
	while(1) {
		TickType_t nowTick = xTaskGetTickCount();
		int txLen = sprintf((char *)buf, "Hello World %"PRIu32, nowTick);
		ESP_LOGI(pcTaskGetName(NULL), "%d byte packet sent...", txLen);

		// Wait for transmission to complete
		if (LoRaSend(buf, txLen, SX126x_TXMODE_SYNC) == false) {
			ESP_LOGE(pcTaskGetName(NULL),"LoRaSend fail");
		}

		// Do not wait for the transmission to be completed
		//LoRaSend(buf, txLen, SX126x_TXMODE_ASYNC );

		int lost = GetPacketLost();
		if (lost != 0) {
			ESP_LOGW(pcTaskGetName(NULL), "%d packets lost", lost);
		}

		vTaskDelay(pdMS_TO_TICKS(1000));
	} // end while
}
#endif // CONFIG_SENDER

#if CONFIG_RECEIVER
void task_rx(void *pvParameters)
{
	ESP_LOGI(pcTaskGetName(NULL), "Start");
	uint8_t buf[256]; // Maximum Payload size of SX1261/62/68 is 255
	while(1) {
		uint8_t rxLen = LoRaReceive(buf, sizeof(buf));
		if ( rxLen > 0 ) { 
			ESP_LOGI(pcTaskGetName(NULL), "%d byte packet received:[%.*s]", rxLen, rxLen, buf);

			int8_t rssi, snr;
			GetPacketStatus(&rssi, &snr);
			ESP_LOGI(pcTaskGetName(NULL), "rssi=%d[dBm] snr=%d[dB]", rssi, snr);
		}
		vTaskDelay(1); // Avoid WatchDog alerts
	} // end while
}
#endif // CONFIG_RECEIVER

void app_main()
{
	gpio_reset_pin(10);
	gpio_set_direction(10, GPIO_MODE_OUTPUT);
	gpio_set_level(10, 1);

	gpio_reset_pin(12);
	gpio_set_direction(12, GPIO_MODE_OUTPUT);
	gpio_set_level(12, 1);

	gpio_reset_pin(39);
	gpio_set_direction(39, GPIO_MODE_OUTPUT);
	gpio_set_level(39, 1);

	// Initialize LoRa
	LoRaInit();
	int8_t txPowerInDbm = 22;

	uint32_t frequencyInHz = 0;
#if CONFIG_433MHZ
	frequencyInHz = 433000000;
	ESP_LOGI(TAG, "Frequency is 433MHz");
#elif CONFIG_866MHZ
	frequencyInHz = 866000000;
	ESP_LOGI(TAG, "Frequency is 866MHz");
#elif CONFIG_915MHZ
	frequencyInHz = 915000000;
	ESP_LOGI(TAG, "Frequency is 915MHz");
#elif CONFIG_OTHER
	ESP_LOGI(TAG, "Frequency is %dMHz", CONFIG_OTHER_FREQUENCY);
	frequencyInHz = CONFIG_OTHER_FREQUENCY * 1000000;
#endif

#if CONFIG_USE_TCXO
	ESP_LOGW(TAG, "Enable TCXO");
	float tcxoVoltage = 3.3; // use TCXO
	bool useRegulatorLDO = true; // use DCDC + LDO
#else
	ESP_LOGW(TAG, "Disable TCXO");
	float tcxoVoltage = 0.0; // don't use TCXO
	bool useRegulatorLDO = false; // use only LDO in all modes
#endif

	//LoRaDebugPrint(true);
	if (LoRaBegin(frequencyInHz, txPowerInDbm, tcxoVoltage, useRegulatorLDO) != 0) {
		ESP_LOGE(TAG, "Does not recognize the module");
		while(1) {
			vTaskDelay(1);
		}
	}
	
	uint8_t spreadingFactor = 7;
	uint8_t bandwidth = 4;
	uint8_t codingRate = 1;
	uint16_t preambleLength = 8;
	uint8_t payloadLen = 0;
	bool crcOn = true;
	bool invertIrq = false;
#if CONFIG_ADVANCED
	spreadingFactor = CONFIG_SF_RATE;
	bandwidth = CONFIG_BANDWIDTH;
	codingRate = CONFIG_CODING_RATE;
#endif
	LoRaConfig(spreadingFactor, bandwidth, codingRate, preambleLength, payloadLen, crcOn, invertIrq);

#if CONFIG_SENDER
	xTaskCreate(&task_tx, "TX", 1024*4, NULL, 5, NULL);
#endif
#if CONFIG_RECEIVER
	xTaskCreate(&task_rx, "RX", 1024*4, NULL, 5, NULL);
#endif
}

```

AkaBkn, 2025-02-10:
yes, it works now! Thanks for your help!

I missed the `gpio_set_direction(x, GPIO_MODE_OUTPUT);`. so I guess the level was never set and thus the SPI errors.

so for conclusion:

```
gpio_reset_pin(BOARD_POWERON);
gpio_set_direction(BOARD_POWERON, GPIO_MODE_OUTPUT);
gpio_set_level(BOARD_POWERON, 1);

gpio_reset_pin(BOARD_TFT_CS);
gpio_set_direction(BOARD_TFT_CS, GPIO_MODE_OUTPUT);
gpio_set_level(BOARD_TFT_CS, 1);

gpio_reset_pin(BOARD_SDCARD_CS);
gpio_set_direction(BOARD_SDCARD_CS, GPIO_MODE_OUTPUT);
gpio_set_level(BOARD_SDCARD_CS, 1);
```

```
CONFIG_MISO_GPIO=38
CONFIG_MOSI_GPIO=41
CONFIG_SCLK_GPIO=40
CONFIG_NSS_GPIO=9
CONFIG_RST_GPIO=17
CONFIG_BUSY_GPIO=13

CONFIG_USE_TCXO=1
```

closed 2025-02-10 by AkaBkn