Xinyuan-LilyGO/T-Deck #79: GPS Taking extremely long to lock or doesn't lock at all

Glenn762702, 2025-04-01:
T-Deck 915MHz Ulbox GNSS

With the factory T-Deck_UnitTest firmware that it came with, I was checking to make sure everything worked well. I noticed the GPS took long to lock and even left it on over night and it still wouldn't lock.  Went outside and it wouldn't lock for about 30 minutes. I drove around and it got a lock. I restarted the device and it wouldn't lock again. I'm not sure what is going on.

I was speaking to someone from the Offical Lilygo Aliexpress store and I was told:

> "Refer to this step instruction ：
> 
> [https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_Reset?spm=5261.27096742.0.0.24f84743CrDO3Z#readme](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_Reset?spm=a2g0s.imconversation.0.0.39813e5f9Id64p#readme)
> 
> flash :
> 
> [https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/GPSShield](https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/examples/GPSShield?spm=a2g0s.imconversation.0.0.39813e5f9Id64p) 
> 
> then use u-center to view the data"

I was confused because the GPSShield.ino is not a .bin file.  Anyways, I followed the instructions and used the webflashing tool [ESP Web Flasher Online](https://espressif.github.io/esptool-js/) and flashed the [GPSShield.ino](https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/examples/GPSShield/GPSShield.ino) file to the T-Deck, and as expected, it would not turn on.  I believe the instructions I'm getting is being lost in translation and I'm not sure what I'm being told what to do exactly.

lewisxhe, 2025-04-02:
Now you can't lock the position, right?

1. Write the factory firmware first. The factory firmware comes with GPS recovery steps. Check whether the position can be locked on the GPS page. If the number of RX columns is cumulative, it means that the GPS output is received and parsed normally.
2. If the position cannot be locked for a long time, you can suspect whether the internal antenna is damaged or the GPS module is damaged. You can find a GPS antenna to replace and test it to see if the position can be locked normally. Normally, the position can be locked in two or three minutes in an open air place. If it still cannot be locked after replacing the antenna, it can be suspected that the GPS module is damaged.

Glenn762702, 2025-04-02:
Just wanted to update that I installed [T-Deck-Plus-Device-UI-v2.6.5.bin](https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/firmware/T-Deck-Plus-Device-UI-v2.6.5.bin) and the GPS appears to lock normally now. I'm not sure why the factory Lilygo firmware was giving me a hard time with the GPS lock.

lewisxhe, 2025-04-07:
Is it difficult to lock the factory firmware?

The factory firmware is the firmware used during T-Deck testing, and it can basically be successfully located within 2 minutes.

closed 2025-05-21 by github-actions[bot]