Xinyuan-LilyGO/T-Deck #82: PlatformIO: Unable to Build UnitTest, FreeRTOS Not Found Errors

kneave, 2025-04-13:
Hi,

I've followed the steps in the readme to open the project in Visual Studio Code and I'm unable to build the UnitTest example.
The errors seem to suggest some dependencies aren't being met but I'm not sure how to install them or what is missing.

Any advice would be greatly received!

```

Processing T-Deck (platform: espressif32@6.3.0; board: T-Deck; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/T-Deck.html
PLATFORM: Espressif 32 (6.3.0) > LilyGo T-Decl(16M Flash 8M OPI PSRAM )
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Converting UnitTest.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ES7210 @ 1.0.0
|-- SensorLib @ 0.2.0
|-- ESP32-audioI2S @ 2.0.6
|-- lvgl @ 8.4.0
|-- RadioLib @ 7.1.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
|-- TFT_eSPI @ 2.5.43
|-- TinyGPSPlus @ 1.0.3
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\T-Deck\src\UnitTest.ino.cpp.o
Compiling .pio\build\T-Deck\src\src\image_emoji.c.o
Compiling .pio\build\T-Deck\src\src\logo_font.c.o
Compiling .pio\build\T-Deck\src\src\wallpaper.c.o
Compiling .pio\build\T-Deck\src\ui.cpp.o
*** [.pio\build\T-Deck\bootloader.bin] Source `C:\Users\keegan\.platformio\packages\framework-arduinoespressif32@3.20009.0\tools\sdk\esp32s3\bin\bootloader_qio_80m.elf' not found, needed by target `.pio\build\T-Deck\bootloader.bin'.
In file included from C:\Users\keegan\AppData\Local\Temp\tmpdbrh2259:1:
C:/Users/keegan/.platformio/packages/framework-arduinoespressif32@3.20009.0/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\T-Deck\src\UnitTest.ino.cpp.o] Error 1
In file included from examples/UnitTest/ui.cpp:10:
C:/Users/keegan/.platformio/packages/framework-arduinoespressif32@3.20009.0/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\T-Deck\src\ui.cpp.o] Error 1
In file included from lib/lvgl/src/misc/../lv_conf_internal.h:15,
                 from lib/lvgl/src/misc/lv_log.h:16,
                 from lib/lvgl/lvgl.h:25,
                 from examples/UnitTest/src/logo_font.c:5:
lib/lvgl/src/misc/../lv_conf_kconfig.h:15:14: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #    include "sdkconfig.h"
              ^~~~~~~~~~~~~
compilation terminated.
In file included from lib/lvgl/src/misc/../lv_conf_internal.h:15,
                 from lib/lvgl/src/misc/lv_log.h:16,
                 from lib/lvgl/lvgl.h:25,
                 from examples/UnitTest/src/image_emoji.c:10:
lib/lvgl/src/misc/../lv_conf_kconfig.h:15:14: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #    include "sdkconfig.h"
              ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\T-Deck\src\src\image_emoji.c.o] Error 1
*** [.pio\build\T-Deck\src\src\logo_font.c.o] Error 1
In file included from lib/lvgl/src/misc/../lv_conf_internal.h:15,
                 from lib/lvgl/src/misc/lv_log.h:16,
                 from lib/lvgl/lvgl.h:25,
                 from examples/UnitTest/src/wallpaper.c:2:
lib/lvgl/src/misc/../lv_conf_kconfig.h:15:14: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #    include "sdkconfig.h"
              ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\T-Deck\src\src\wallpaper.c.o] Error 1
================================================================== [FAILED] Took 10.72 seconds ==================================================================

Environment    Status    Duration
-------------  --------  ------------
T-Deck         FAILED    00:00:10.723
============================================================= 1 failed, 0 succeeded in 00:00:10.723 ============================================================= 

```

lewisxhe, 2025-04-14:
This error should be that your platformio esp32 package is not installed completely, delete .platformio and reinstall it

kneave, 2025-05-15:
> This error should be that your platformio esp32 package is not installed completely, delete .platformio and reinstall it

That worked, thanks!

closed 2025-05-15 by kneave