Xinyuan-LilyGO/T-Deck #83: GPS location is wrong

marai0162, 2025-04-26:
hy are the latitude and longitude values ​​off by about 800 meters even though eight satellites have been captured?
The same issue occurs with both meshtastic versions 2.6.4 and 2.60.
Furthermore, the GPS antenna that arrived was upside down, so this issue occurred after I corrected it.
One antenna was in poor condition and had difficulty picking up satellite signals, so we replaced it with another antenna.

lewisxhe, 2025-04-28:
What is the effect after replacing the antenna?

OneSleeve, 2025-05-15:
I'm having very similar issues with the t-deck plus with the 2.6.4 firmware. It's also updating very very infrequent regardles of what i set in the settings.

OneSleeve, 2025-05-16:
Issue also seems to persist in 2.6.8 alpha.

patdemko, 2025-06-26:
I'm seeing a very similar issue with a tdeck I just bought.  I'm running meshtastic 2.6.11 and not using an external antenna.    It's seeing 12 sats, but the location is off 345 meters according to the device.   I've also traveled a bit with it and it seems to consistently be off about 200 to 300 meters to the southwest.

closed 2025-08-10 by github-actions[bot]