Xinyuan-LilyGO/T-Deck #85: TFT display datasheet / problem

omikron88, 2025-05-29:
Hello,
May you public the datasheet of the TFT display used currently?
I am still having some troubles in Arduino IDE. It reboots during tft.begin()

omikron88, 2025-05-30:
> If it restarts, please check your pin configuration. Please use the
> TFT_eSPI copy provided in this repository to avoid configuring TFT_eSPI.

Yes, but it still reboots in begin, even if provider eTFT copied into
library folder.

lewisxhe, 2025-06-03:
Restarting when calling tft.begin only happens when the pin assignment is wrong. Are you sure that restarting occurs when calling tft.begin ? What if you do something else instead of calling tft.begin ? For example, print hello ?

closed 2025-07-17 by github-actions[bot]