Xinyuan-LilyGO/T-Deck #89: I think my tdeck is defective and your support chat won't work

framerate, 2025-06-16:
Order LILYGO-47827

My TDeck cannot connect to any nodes. I have a heltec v3 here connecting to tons with a smaller antennae. I tested with the included antennae and a much larger one and I cannot connect to any nodes. I think maybe the antennae connector or radio chip is defective.

I've also flashed the latest meshtastic firmware so I've ruled out its a meshtastic problem and appears to be a hardware problem

Please let me know how to file a RMA.

lewisxhe, 2025-06-18:
Here, select the T-Deck_UnitTest_241220.bin file and write it according to the method in README. Switch RX and TX in the radio interface. Normally, TX should be valid in the state of a single node, and RX requires two nodes with the same settings.
https://github.com/Xinyuan-LilyGO/T-Deck/tree/master/firmware

lewisxhe, 2025-06-21:
It looks like the radio is working fine. I even suspect that LilyGo sent you the wrong frequency. The frequency should be marked on your packaging shell and antenna. What is written on it?

You can also remove the shell of the t-deck, and the radio module inside also has the frequency marked on it.

framerate, 2025-06-23:
Its not the plus, so its the board only. The radio is marked 868/915mhz, which I believe is correct? I have an external antennae attached as well for 915Mhz.

Please advise, I don't know how to reach out other than this github thread to get this replaced. (PS I have a TDeck Plus coming today or tomorrow so I can compare reception as well)

framerate, 2025-06-30:
@lewisxhe I solved this problem. On the Meshtastic discord, they told me there's a known issue with the latest firmware(s) that its not setting the default channel when you flash the firmware.

I think its setting channel 0 and you have to set channel 20 manually. Then it starts to work (in case someone else reports this)

closed 2025-06-30 by framerate

lewisxhe, 2025-07-01:
Yes, someone has reported this issue to me. Thanks for your answer.