Xinyuan-LilyGO/T-Deck #92: Still Cant Rx or TX

MikeMesh, 2025-07-09:
Greetings,

fired up a brand new Tdeck plus today. Wont talk to other nodes. Discovered the channel issue, changed slot to 20, freq moved to 906.875 as its supposed to. Reflashed firmware, factory resets, nothing. Still no go. 

But none of that solved the issue. I have many other nodes so I am knowledgeable and familiar with meshtastic, but I cant make this one talk. 

I can see it transmitting on a spectrum analyzer but nothing understands it and it understands no other nodes. 

Ive tried the public channel, custom channels, removing encryption, but nothing works. 

Any advice would be greatly welcomed. 

Cheers

MikeMesh, 2025-07-09:
I cant guess whats going on. Yes my problem is similar to that of 89, but changing the freq slot to 20 did not solve it. Nor did reflashing of resetting the firmware. I even back up to one older firmware and tried it. Still no go. Ive done just about everything I can think of. I am starting to suspect there may be an issue with the sx1262 radio. I have tried every setting. I so wanted this to go. This is a great looking package!

lewisxhe, 2025-07-09:
OK! I need @mverch67 's advice

MikeMesh, 2025-07-09:
Thank you. Its 11:30 at night here and i got work tomorrow. So im off to get some sleep. I shall return around 7 pm EST. 

Cheers and thank you again.

MikeMesh, 2025-07-09:
oh, everything else works fine. the screen, the menus etc. I see no issues here.

mverch67, 2025-07-09:
@MikeMesh 
> I can see it transmitting on a spectrum analyzer but nothing understands it and it understands no other nodes.

How do you determine _"nothing understands it"_? Did you look into the USB serial debug log? When sending and receiving you always see detailed information that clearly gives you a hint what went wrong, e.g. here a packet is received but cannot be decoded:

```
DEBUG | ??:??:?? 77 [RadioIf] Lora RX (id=0x39dc4642 fr=0x17173035 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x1e encrypted len=50 rxSNR=6.75 rxRSSI=-65 hopStart=7 relay=0xdc)
DEBUG | ??:??:?? 77 [RadioIf] Packet RX: 608ms
INFO  | ??:??:?? 77 [Router] Packet History - insert: Using new slot @uptime 77.676s TRACE NEW
WARN  | ??:??:?? 77 [Router] No suitable channel found for decoding, hash was 0x1e!
DEBUG | ??:??:?? 77 [Router] packet decoding failed or skipped (no PSK?) (id=0x39dc4642 fr=0x17173035 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x1e encrypted len=50 rxSNR=6.75 rxRSSI=-65 hop
```

MikeMesh, 2025-07-09:
[meshtastic_debug_20250709_070405.log](https://github.com/user-attachments/files/21141157/meshtastic_debug_20250709_070405.log)

See the attached file. the unit does not see other nodes, other nodes do not see it.

mverch67, 2025-07-10:
LoRa works fine! This is your issue:

`WARN | 00:10:56 141 [Router] No suitable channel found for decoding, hash was 0xa!`

mverch67, 2025-07-10:
> They are matched. Perfectly. AQ==

Do you have proof? Screenshot, export-config, etc. will do.

mverch67, 2025-07-10:
What device is that? The T-Deck? What about the other device?

MikeMesh, 2025-07-10:
> What device is that? The T-Deck? What about the other device?

Yes, all the other devices are EXACLTY the same. All the other devices are communicating. The screen shot I sent you is the T deck plus, the problem device. 
They are ALL the same. 7 of them. All in close proximity. All work except the Tdeck plus

ALL LongFast AQ==

mverch67, 2025-07-10:
Did you type in the "LongFast"? Usually for the default channel it should be kept blank, so that when changing the preset (e.g. to MediumFast) the UI can replace the channel name according to the chosen preset.

MikeMesh, 2025-07-10:
> Did you type in the "LongFast"? Usually for the default channel it should be kept blank, so that when changing the preset (e.g. to MediumFast) the UI can replace the channel name according to the chosen preset.

I have tried this no less than 3 ways several times.

1) I went with what the firmware was preprogrammed with for primary channel. Didnt work. I have never had a preprogrammed primary channel fail to work till now. 
2) I re-entered manually LongFast and AQ==. Didnt work.
3) I left the primary channel blank with AQ== psk. Didnt work.
4) I entered my private channel and psk, which operates and shared with many other nodes and for the T deck plus, it didnt work. 
5) when i make these changes, I reboot the device and ensure my settings were in place. I have programmed over 40 of these nodes, so I have some fairly good experience in doing so. 
6) I have tried the above with no encryption. It didnt work. 

Notes.

All of the above has been tried and check several times with no results. I reinstalled the firmware thinking it might have been a bad flash. 
I went thru the above drill again. It didnt work.
Keep in mind in the LoRa section, I had to change the slot number to 20 and I also ensured the unit was transmitting centered on 906.875.
I installed older firmware and did all the tests again. It didnt work. I also checked and ensured the slot freq was correct.
Since then I have reinstalled the latest firmware, changed the slot to 20 and tried the above again. It didnt work.

MikeMesh, 2025-07-10:
> What you could try is use a different preset with a lower spreading factor (SF) to see if it's the LoRa radio that has difficulties at SF11 (see https://meshtastic.org/docs/overview/radio-settings/#presets). I.e. set the T-Deck and another device to e.g. MediumFast and then see if they hear each other or still have this decode issue.

I set up a test to do this. 

1)Installed fresh firmware on the Tdeck Plus. Configured the Region, gave it a fast name and changed the slot freq to 20. Confirmed transmit at 906.875.  Added my private channel and psk as well. Two channels showing. The primary LongFast and my private channel. Confirmed all psks are correct.
2) set the modem to Long_Slow.
3) I set one V3, One T114 and a RAK 19007 all to Long_Slow. They are share the same channels and I tested them all at this speed rate. All of these radios listed here connected to each other without issue.
4) The Tdeck did not connect. It did not work.

MikeMesh, 2025-07-11:
> OK. I have seen the above message. I apply to send you a new device. As for the device in question, let's see if it can be sent back to us. I have seen your message on the customer service platform.

@lewisxhe 

Thank you. I cant stress how important this is. It all needs to happen very fast.

mverch67, 2025-07-11:
@MikeMesh
> 3) I set one V3, One T114 and a RAK 19007 all to Long_Slow. They are share the same channels and I tested them all at this speed rate. All of these radios listed here connected to each other without issue.

LongSlow is the wrong test because it is SF12. If the radio has problems with SF11 it will probably also not work on SF12. My request was to test lower spreading factors (see https://meshtastic.org/docs/overview/radio-settings/#presets) like MediumFast (SF9).

Then, can you also test with the latest firmware. It solved the channel 1 -> 20 issue. Then only set region and give it short/long name in the initial popup dialog and reboot. No further channel settings, keep all default(!).

MikeMesh, 2025-07-11:
@mverch67 

I see no new firmware. Only 2.6.11. This is what I  am running. It has the error in it. 

I do not have Python CLI set up yet. Its on my list of things to do, but not today. 

[image]

MikeMesh, 2025-07-11:
> But it has successfully sent a telemetry packet. What did your other nodes do when they received this packet?
> 
> ```
> DEBUG | ??:??:?? 67 [RadioIf] Started Tx (id=0x94df4b9c fr=0xa0a9ed78 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=48 hopStart=3 relay=0x78 priority=10)
> DEBUG | ??:??:?? 67 [RadioIf] Packet TX: 591ms
> ```

The other nodes dont see it or show in their respective nodes list. its been running for hours now.

mverch67, 2025-07-11:
> The other nodes dont see it or show in their respective nodes list. its been running for hours now.

I mean what do they write into their debug log?

mverch67, 2025-07-11:
Do they write this line (**hash was 0x8**) or what is their hash code they'd complain about?

`WARN | ??:??:?? 47 [Router] No suitable channel found for decoding, hash was 0x8!`

mverch67, 2025-07-11:
> Heres a look at the log, same error. I fully believe there is a hardware issue with the radio.

Not very likely because all other packet data looks correct (and the chip has error correction) except the hash code. Which lets assume the error is due to different channel config at sending/receiving node.

MikeMesh, 2025-07-11:
DEBUG | 18:05:59 60085 [Router] packet decoding failed or skipped (no PSK?) (id=0x6a5946ba fr=0xa0a9ed78 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=48 rxtime=1752257159 rxSNR=6.

[image]

mverch67, 2025-07-11:
I just checked on my T-Deck: I've set it to US/20/906.875 and see the following in the log:

```
DEBUG | ??:??:?? 166 [RadioIf] Started Tx (id=0x9f915589 fr=0x99bc689c to=0xffffffff, WantAck=0, HopLim=3 **Ch=0x8** encrypted len=113 hopStart=3 relay=0x9c priority=64)
DEBUG | ??:??:?? 166 [RadioIf] Packet TX: 1075ms
DEBUG | ??:??:?? 166 [RadioIf] 0 packets remain in the TX queue
DEBUG | ??:??:?? 167 [RadioIf] Completed sending (id=0x9f915589 fr=0x99bc689c to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=113 hopStart=3 relay=0x9c priority=64)
```

You see: **Ch=0x8** 

That means your T-Decks' behavior is same as mine and its configuration is correct. The default LongFast channel hash is **0x08**. Now the question is why your other nodes are sending with a different channel hash **0x0a** ??

mverch67, 2025-07-11:
This **0x0a** could be the your second encrypted channel on your other devices. But if they have another <empty name> LongFast channel with PSK AQ== then they would be able to decode the messages from your T-Deck. It seems they have not.

mverch67, 2025-07-11:
> -The T114 did not see the T Deck. The T Deck did not see the T114.

It takes a while to get other nodes' packets (especially in 2.7.x). Can take 15mins or longer. To speed up you can use MUI's Mesh Detektor. Then it takes only minutes.

MikeMesh, 2025-07-11:
> > -The T114 did not see the T Deck. The T Deck did not see the T114.
> 
> It takes a while to get other nodes' packets (especially in 2.7.x). Can take 15mins or longer. To speed up you can use MUI's Mesh Detector. Then it takes only minutes.

Its been hours now. The T Deck is not being received. 

Manuel, Thank you for your help. I officially give up on this one. Geez, it just crashed again. Wont respond until you turn it off then back on again. 

Cheers

MikeMesh, 2025-07-12:
> I have already talked to the dealer and asked if we can send your device back to us for analysis. We have arranged for a new device to be sent to you. You should have received a reply from the dealer.

I have not heard from the dealer nor have I received a tracking number. I sure hope we make this in less than 2 weeks. Has it shipped?

lewisxhe, 2025-07-13:
Yes. I have already asked about it and they have sent it to the warehouse for shipping.

MikeMesh, 2025-07-20:
@lewisxhe The new unit arrived today and is working well. no issues at this point. I thank you very kindly. I appreciate your help very much. 

While I was waiting for this replacement, I ordered another thru your web store just to be safe. It arrived a few days ago. It works fine on the network, but the battery is hard pressed to last 4 to 6 hours. 

I dont care about warranty at this point as my presentation grows close. 

Can you tell me-

1) Where can I buy a few spare batteries for these units?
2) Is there a larger capacity battery that I can buy, say 2500 or 3000 mah?

Thank you.

mverch67, 2025-07-20:
> Is there a larger capacity battery that I can buy, say 2500 or 3000 mah?

[image]

MikeMesh, 2025-07-20:
@mverch67 

Got it. Thanks so 505070 is max size correct?

lewisxhe, 2025-07-22:
Yes.support