Xinyuan-LilyGO/T-Deck #94: New T-Deck does not see other nodes or communicate to other nodes

tcanthony, 2025-07-15:
I have updated the new T-Deck to 2.6.1. It seems to run without rebooting and I have gotten the GPS to work. What I can't do is detect any of my other nodes, nor can I send a message to any of my other nodes through messaging. I have confirmed the name of the channel and the preshared key is exactly the same. 
On all of my other units, the primary channel is just default with AQ== as the shared key. I have 6 other units of varying brands and they all communicate via the primary channel and see other nodes in my area. I have a secondary channel with a specific key that all of my other units can communicate with. The T-Deck will not. I have created a new channel on the T-Deck first and then tried to set one up on an older unit but still no communications. 
All of the other units just "worked" when I flashed them. This T-Deck seems unable to do anything except read the GPS.
Any ideas?

tcanthony, 2025-07-15:
It also says "No Signal" on the home screen. Unlike what I have seen on other units.

[image]

lewisxhe, 2025-07-16:
This problem has been popping up recently. Please check these two issues to see if they are the same problem.
https://github.com/Xinyuan-LilyGO/T-Deck/issues/92
https://github.com/Xinyuan-LilyGO/T-Deck/issues/89

lewisxhe, 2025-07-16:
@mverch67 Recently, some people have reported this kind of problem. What do you think?

Can I flash back to the old version without UI for testing?

mverch67, 2025-07-16:
@tcanthony

To be able to communicate with other radios make sure they all are running on the **exact same** frequency. In your error report you didn't mention what frequency you've setup. It's displayed on the home screen of the T-Deck and can be changed via Settings -> Modem Preset -> Frequency Slot.

closed 2025-07-16 by tcanthony