Xinyuan-LilyGO/T-Deck #97: T Deck Plus Backlight on but nothing appears

Zoxs97, 2025-08-02:
I have tried flashing every single example from the repository, sometimes the backlight is on, sometimes its off but in each case nothing display to the screen. Is there diffferent config for T Deck Plus? do i need to change something in User_Setup.h ? I've looked online tried everything but nothing ever appears

In my user_setup_select.h i have this selected #include <User_Setups/Setup210_LilyGo_T_Deck.h>

and this is my setup for the Setup210_LilyGo_T_Deck.h 
`#define USER_SETUP_ID 210
#define ST7789_DRIVER

#define TFT_WIDTH  320
#define TFT_HEIGHT 240

#define CGRAM_OFFSET
#define TFT_RGB_ORDER TFT_RGB
#define INIT_SEQUENCE_2
#define TFT_INVERSION_ON

#define TFT_MISO 38
#define TFT_MOSI 41
#define TFT_SCLK 40
#define TFT_CS   -1
#define TFT_DC   11
#define TFT_RST  -1
#define TFT_BL   42
#define TFT_BACKLIGHT_ON 1
#define BOARD_POWERON       10
#define BOARD_TFT_BACKLIGHT 42

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

#define SPI_FREQUENCY       40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY 2500000
`

lewisxhe, 2025-08-05:
Please copy the lib directory to arduino libraries as is. Do not upgrade and use esp32 core version 2.0.14.

Zoxs97, 2025-08-05:
Thank you I was on version 3.0.3, it fixed it. Thanks a lot!

closed 2025-08-05 by Zoxs97